Xinyuan-LilyGO/Wrist-E-Paper #3: Timed out waiting for packet header

martin-gk, 2021-12-11:
Hi,
after succesfully testing the basic examples I uploaded a more complicated sketch to the board. Everything works fine except I cannot upload anything anymore. I always get the timed out waiting for packet header error. I've tried to use the boot button as well as the reset button. Could you offer me some assistance.

LilyGO, 2022-02-11:
I guess it is caused by the poor contact of U2T. It is recommended that you try to read the serial port information, and press the reset button repeatedly, until the serial port returns information, which proves that the communication is normal, and then upload the program.

LilyGO, 2022-08-02:
The problem is too long, you can open it again if necessary

closed 2022-08-02 by LilyGO