Xinyuan-LilyGO/Wrist-E-Paper #4: T-U2T with Wrist-E-Paper Eval kit not working?

jabadabado, 2022-01-17:
Hi, I bought a LilyGo Dev module online, the T-U2T is a USB to Uart TTL converter right? It is completely dead. 
Plugging in the USB plug straight into the ESP32 board makes the computer complain that it is not recognizing the USB port. How do I get this USB Adapter to work?

LilyGO, 2022-02-11:
I guess it is caused by the poor contact of U2T. It is recommended that you try to read the serial port information, and press the reset button repeatedly, until the serial port returns information, which proves that the communication is normal, and then upload the program.

Remember to install the U2T driver, which is essentially a serial port conversion chip. CH340K or 2104

LilyGO, 2022-08-02:
The problem is too long, you can open it again if necessary

closed 2022-08-02 by LilyGO