Xinyuan-LilyGO/Wrist-E-Paper #6: Programming E-Paper Watch through GPS connector

mrdROb, 2022-12-15:
I accidentally bought the device without T-U2T adapter, and while I am waiting for one, want to give the board a try
Can I program the board through the GPS connector? Are any drawbacks?

mmMicky, 2022-12-16:
If you have not purchased U2T, you need a serial communication tool and a TYPE-C cable. Only need to connect TX and RX. You also need to enter download mode manually.

mrdROb, 2022-12-18:
Thanks a lot! After looking more at the schematic I found that I was wrong identifying that GPS connector/soldering pads have RX and TX. I will go the way you recommend. Since I now have a full USB Type-C connector, I will also try to replicate the BOOT and RESET lines from my CH340 programmer.

closed 2022-12-18 by mrdROb