Xinyuan-LilyGO/Wrist-E-Paper #7: Backlight?

qewer33, 2023-07-05:
I was wondering if this board really has a backlight or not. The reference image in the store product page mentions IO33 as "Backlight" but below code does not seem to work:
```cpp
#define BACKLIGHT 33

pinMode(BACKLIGHT, OUTPUT);
digitalWrite(BACKLIGHT, HIGH);
```
Does this board not have a backlight or am I doing something wrong? Thanks in advance!

alainrx, 2024-03-24:
There is a connector in order to connect a backlight. But the backlight itself is not included.

alex-sugden, 2024-05-23:
Do you know anywhere one might find a compatible one?

alainrx, 2024-05-23:
> Do you know anywhere one might find a compatible one?

It seems that the connector is compatible with the "front-light" GooDisplay (small flat cable). The e-paper display + front-light is available on Aliexpress : https://fr.aliexpress.com/item/1005006277614640.html  (not tested)