Xinyuan-LilyGO/Wrist-E-Paper #8: Free GPIO?

JanB97, 2023-10-18:
Are there any free gpio pins left for connecting a temperature/humidity sensor?

alainrx, 2024-01-06:
I have the same question, interested to use this e-paper for thermometer. So looking to the schematic, with the version without GPS ship, it should be possible to use the GPIO pins going to the GPS chip (GPIO21, 22 -> I2C, GPIO18, 19, 23). Very small soldering of wires to be done...
I have ordered one, I will try.

JanB97, 2024-01-11:
I have settled for a way cheaper option for what I wanted to do with it, but I am still interested for your results.

alainrx, 2024-02-10:
I confirm that it is possible to use I2C pins of the ESP on the version without GPS ship. Just sold wires at the placement of D9 (SCL : GPIO22) and D10 (SDA : GPIO21). You can also find power supply V3V (which is controlled by PWR_EN) and 0V. See picture. I have connected HDC1080 with success.
[image]

alainrx, 2024-02-10:
This is my prototype of thermometer with the sensor. The battery voltage measurement is not at all accurate with the ADC of the ESP32, so correction factor have to be done.
[image]